nakhonthai/ESP32APRS_LoRa #1: [Question] compatibility with other LoRa APRS Tracker and iGate software

lyusupov, 2024-04-10:
Thank you for sharing!

Is this new software compatible in **packet data format** and **radio frequency being used** with existing LoRa Tracking and iGate software, like :

1) [Tracker](https://github.com/lora-aprs/LoRa_APRS_Tracker) and [iGate](https://github.com/lora-aprs/LoRa_APRS_iGate) from Peter Buchegger ?
2) [Tracker](https://github.com/richonguzman/LoRa_APRS_Tracker) and [iGate](https://github.com/richonguzman/LoRa_APRS_iGate) from Ricardo Guzman ?

nakhonthai, 2024-04-10:
Default by use AX.25 ,but it can received from tracker and digi from LoRa APRS other.
Use as IGate received from LoRa APRS other , OK.
Use as Digi/Tracker send to LoRa APRS other. Not OK.

lyusupov, 2024-04-10:
@nakhonthai 

1) the first tracker operates @ 433.775 MHz and uses a fixed 3-bytes prefix ahead of  AX25 payload:
https://github.com/lora-aprs/LoRa_APRS_Tracker/blob/master/src/LoRa_APRS_Tracker.cpp#L314-L316

2) the second tracker declares a certain compatibility mode with the first one (@ 433.775 MHz) . However, it also has a implementation of it's own `LoRa APRS438` protocol @ 438 MHz, that is explained there: https://aprs438.readthedocs.io/en/latest/index.html

herbsens, 2024-04-15:
Hi, thank you for this project. I want to know about making an Igate with only GPS and OLED screen functions, is it possible? If possible, how should I go about selecting and setting up the PlatformIO Project Configuration

herbsens, 2024-04-15:
> Default by use AX.25 ,but it can received from tracker and digi from LoRa APRS other. Use as IGate received from LoRa APRS other , OK. Use as Digi/Tracker send to LoRa APRS other. Not OK.

Hi, thank you for this project. I want to know about making an Igate with only GPS and OLED screen functions, is it possible? If possible, how should I go about selecting and setting up the PlatformIO Project Configuration

closed 2024-04-15 by lyusupov