nakhonthai/ESP32APRS_LoRa #5: Heltec Wireless Tracker Battery Voltage Sensor

PorterRyan, 2024-07-13:
It would be nice to have some instructions on adding the battery voltage as a sensor. I've fiddled with the software but the voltage always outputs as 0.00V.

Szetya, 2024-07-14:
[image]
@nakhonthai  The ADC_CTRL PIN must be set to HIGH before measurement. 🙏😇

nakhonthai, 2024-07-21:
OK , for Heltec HTIT-Tracker fix next version 0.2c coming soon.

nakhonthai, 2024-07-21:
> ![image](https://private-user-images.githubusercontent.com/22783982/348516574-fdd6990b-ee7e-4da6-b2f0-380f4a56dac8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Warning! ADC_Ctl1 on B pin transistor connected direct to GPIO(forward diode). It use input pullup mode for GPIO.

closed 2024-07-21 by nakhonthai

Szetya, 2024-07-29:
> OK , for Heltec HTIT-Tracker fix next version 0.2c coming soon.

I made a few changes. 
Please see the pull request. 😇

nakhonthai, 2024-08-04:
Update V0.2c now.

Szetya, 2024-08-04:
> Update V0.2c now.

Thank you! :)

/off
(there were some translation warnings, could you check my commits? https://github.com/nakhonthai/ESP32APRS_LoRa/pull/8/commits
Less confusing with the warnings and a better solution for the wdt.h directive (S3 doesn't need wdt.h)
/on