nakhonthai/ESP32APRS_LoRa #6: Lora failed

samsamxu, 2024-07-16:
Dear Nakhonthai,
I flashed the esp32s3 and mod the lora, the lora  Init success, but when sending message, it shows [LoRa] Failed, code -24. 
Please check the code as follows: 
`
[1017000][D][main.cpp:4061] loop(): Free heap: 198.0 KB         WiFi:Mate 60 Pro ,RSSI:-50 dBm
[1022470][D][LibAPRS.cpp:183] APRS_init(): [SX12xx] Initializing .. 
[1022470][D][LibAPRS.cpp:186] APRS_init(): Init chip SX1278
[1022471][D][LibAPRS.cpp:261] APRS_init(): [9] Begin... 
[1022489][D][LibAPRS.cpp:269] APRS_init(): [LoRa] Init success!
[1022490][D][LibAPRS.cpp:283] APRS_init(): [LoRa] Start receive success!
Client reconnected! Last message ID that it got is: 732458
[1027005][D][main.cpp:4060] loop(): Task process APRS=40uS       NETWORK=4080uS  GPS=17uS        SERIAL=0uS      TempCPU=0.00

[1027006][D][main.cpp:4061] loop(): Free heap: 195.1 KB         WiFi:Mate 60 Pro ,RSSI:-45 dBm
[1037007][D][main.cpp:4060] loop(): Task process APRS=53uS       NETWORK=4147uS  GPS=27uS        SERIAL=0uS      TempCPU=0.00

[1037008][D][main.cpp:4061] loop(): Free heap: 195.1 KB         WiFi:Mate 60 Pro ,RSSI:-40 dBm
[1047008][D][main.cpp:4060] loop(): Task process APRS=28uS       NETWORK=4078uS  GPS=14uS        SERIAL=0uS      TempCPU=0.00

[1047009][D][main.cpp:4061] loop(): Free heap: 195.8 KB         WiFi:Mate 60 Pro ,RSSI:-34 dBm
[1057009][D][main.cpp:4060] loop(): Task process APRS=37uS       NETWORK=4075uS  GPS=13uS        SERIAL=0uS      TempCPU=0.00

[1057010][D][main.cpp:4061] loop(): Free heap: 195.8 KB         WiFi:Mate 60 Pro ,RSSI:-49 dBm
[1067010][D][main.cpp:4060] loop(): Task process APRS=40uS       NETWORK=4080uS  GPS=12uS        SERIAL=0uS      TempCPU=0.00

[1067011][D][main.cpp:4061] loop(): Free heap: 195.8 KB         WiFi:Mate 60 Pro ,RSSI:-67 dBm
[1076547][D][main.cpp:5744] taskAPRS(): TRACKER RAW: BI9BCQ-7>APE32L-2:>  ?RSSI TRACKER MODE

[1076548][D][main.cpp:5745] taskAPRS(): TRACKER EVENT_TX_POSITION=4      INTERVAL=60

[1076562][D][main.cpp:2630] pkgTxSend(): TX->RF: BI9BCQ-7>APE32L-2:>  ?RSSI TRACKER MODE
[1077676][D][main.cpp:4060] loop(): Task process APRS=601uS      NETWORK=4071uS  GPS=16uS        SERIAL=0uS      TempCPU=0.00

[1077677][D][main.cpp:4061] loop(): Free heap: 195.5 KB         WiFi:Mate 60 Pro ,RSSI:-74 dBm
[1079536][D][LibAPRS.cpp:122] radioRecvStatus(): [LoRa] RSSI:-164 dBm   SNR:0 dBm       FreqErr:0 Hz
[1079537][D][LibAPRS.cpp:313] APRS_poll(): [LoRa] Received packet! 0 Byte

[1079541][D][LibAPRS.cpp:362] APRS_poll(): [LoRa] Failed, code -24
Client reconnected! Last message ID that it got is: 1025428
[1087677][D][main.cpp:4060] loop(): Task process APRS=30uS       NETWORK=4076uS  GPS=17uS        SERIAL=0uS      TempCPU=0.00`

nakhonthai, 2024-07-21:
Received error from noise. `radioRecvStatus(): [LoRa] RSSI:-164 dBm SNR:0 dBm FreqErr:0 Hz`
` [LoRa] Received packet! 0 Byte`

samsamxu, 2024-07-22:
Thank you, I reconnect the wires, it works. 
By the way, how to connect the 4G LTE? 
I can not find where and how to connect this LTE modem