nakhonthai/ESP32APRS_LoRa #11: Comunication with IGate

DK7AT, 2024-10-06:
The Tracker firmware on The Heltech HTIT Board is sending packages but my LilyGo IGate (Richon Guzman Firmware) wont receive/hear them.
I can hear that the tracker sends with my Handheld Radio.

can you help me?

73 de Tim DK7AT

nakhonthai, 2024-10-07:
This firmware use AX.25 not support send data to other firmware.
But, it can received from other firmware.

DL7AIR, 2024-10-30:
Sawadee Nakhonthai, I have the same problem as Tim. My HTIT is sending packages, but the next Lora IGate DB0PSW didn't receive/decode my packages. Everything else is fantastic, you did a very good job.

73 de Klaus, DL7AIR

DK7AT, 2024-10-31:
Hallo Klaus. 
du musst auch die Firmware von Ihm auf dem IGate haben. Seine TX Firmware funktioniert mit anderen RX Firmwares nicht. andersrum gehts. Ich nutze ein Heltec V3 Board mit seiner Firmware als IGate. Tracker im Auto ist mit der Firmware von Richon Guzman versehen. Mein Portabel Tracker mit der Firmware von Nakhonthai. Wird beides von seiner Firmware gehört.

73 de Tim DK7AT

DL7AIR, 2024-10-31:
Hallo Tim,
jetzt verstehe ich, wie das gemeint ist. Ich rätsel schon seit Tagen, warum das IGate mich nicht lesen kann. Schade, denn ich finde die Firmware von Nakhonthai eigentlich ganz gut. Besunders die Unterstützung für das Farbdisplay und der Webserver. Vielen Dank für den Hinweis. 73 und Happy Halloween de Klaus, DL7AIR

DK7AT, 2024-10-31:
kein problem,

ich hatte vorher ein TTGO Board als IGate, da funktioniert seine Firmware (noch) nicht drauf, daher die umstellung auf das Heltec Board.

Ganz wichtig, wenn du den HTIT mit nem Akku nutzt, Löte einen Schalter dazwischen um die spannung zu unterbrechen, da das Gerätchen dir sonst den akku leer zieht. Oder ich habe nicht verstanden wie man das Ding "richtig" ausschaltet ;)
Ob es an der Firmware oder am Gerät liegt, konnte ich noch nicht ergründen. 

LG

DL7AIR, 2024-11-01:
Hi Tim,
ich habe mir gerade die Firmware von richon guzman geflasht. In der Konfiguration gibt es den Punkt "Monitor Battery Voltage and power-off/sleep if low" Vielleicht löst das ja das problem. 73 de Klaus

lyusupov, 2024-11-18:
duplicates https://github.com/nakhonthai/ESP32APRS_LoRa/issues/1

nakhonthai, 2025-08-25:
Last version supported by switch off AX.25 in the tab RADIO

closed 2025-08-25 by nakhonthai